robotology/icub-models #189: Cannot find all robot models after #186

xEnVrE, 2023-02-15:
I have just installed the latest `devel` of this repository via the `robolology-superbuild` using `ROBOTOLOGY_PROJECT_TAGS = Unstable`.

The installation is from scratch, hence the `install` folder of the superbuild did not contain any leftovers.

As of now, when I run `gazebo` I can only see a partial list of the models:

[image]

and running with `gazebo --verbose` tells that:

```console
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubDarmstadt01"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubErzelli02"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubGazeboV2_5"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubGazeboV2_5_plus"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubGazeboV2_6"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubGazeboV2_7"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubGazeboV3"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubGenova01"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubGenova02"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubGenova03"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubGenova04"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubGenova04_plus"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubGenova09"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubLisboa01"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubNancy01"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubParis01"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubParis02"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/ICUBcontrib"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/YCM"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/ament_index"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/bash-completion"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/cmake"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/ergoCub"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/robotology-superbuild"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/yarp"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/ergoCub/robots/ergoCubSN000"
```

Reverting back #186, I get back the usual list of robots:

[image]

and `gazebo --verbose` reports:

```console
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubDarmstadt01"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubErzelli02"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubGazeboV2_6"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubGazeboV2_7"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubGenova01"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubGenova02"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubGenova03"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubGenova04"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubGenova04_plus"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubGenova09"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubLisboa01"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubNancy01"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubParis01"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub/robots/iCubParis02"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/ICUBcontrib"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/YCM"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/ament_index"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/bash-completion"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/cmake"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/ergoCub"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/iCub"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/robotology-superbuild"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/yarp"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/home/user/robotology-superbuild/build/install/share/ergoCub/robots/ergoCubSN000"
```

cc @traversaro

traversaro, 2023-02-15:
Thanks @xEnVrE for the issue, https://github.com/robotology/icub-models/pull/190 should fix the problem. Anyhow, this indicates how we are fragile, it would be great to have some CI tests that at least test that the models are loaded correctly in Gazebo.

traversaro, 2023-02-15:
Fixed in https://github.com/robotology/icub-models/pull/190 .

closed 2023-02-15 by traversaro