robotology/icub-models #191: The order of torso joints is wrong in latest devel

xEnVrE, 2023-02-16:
According to the [official documentation](https://icub-tech-iit.github.io/documentation/icub_kinematics/icub-joints/icub-joints/#torso), the correct order of the torso joints for the `iCub 2.x` should be:

[image]

i.e., `torso_yaw`, `torso_roll` and then `torso_pitch`.

`icub-models v1.26.0` enforces the correct order:

https://github.com/robotology/icub-models/blob/6e3a79d8520d74e549716e66f2f549e9e2cf3817/iCub/conf/gazebo_icub_torso.ini#L27

However, in the latest devel, `torso_roll` and `torso_pitch` are swapped:

https://github.com/robotology/icub-models/blob/f90f0d60c1e461133bfa2638fa689de1635daec8/iCub/conf/gazebo_icub_torso.ini#L3

This is breaking all the software making use of forward kinematics routines.

Is this intended or is it just an error?

xEnVrE, 2023-02-16:
cc @traversaro

xEnVrE, 2023-02-16:
It seems the source of the error is here:

https://github.com/robotology/icub-models-generator/commit/c9e5e08fcfecbce48138662d891d9131fe3e3908#diff-1dc880f789fa7a9b157372ceb878d57d08af4ed374e5a50daba6ae9578ef3c4b

traversaro, 2023-02-16:
I think it is an error, feel free to open a PR to fix it, I will do it once I am in front of the laptop. fyi @Nicogene

Nicogene, 2023-02-16:
> It seems the source of the error is here:
> 
> [robotology/icub-models-generator@c9e5e08#diff-1dc880f789fa7a9b157372ceb878d57d08af4ed374e5a50daba6ae9578ef3c4b](https://github.com/robotology/icub-models-generator/commit/c9e5e08fcfecbce48138662d891d9131fe3e3908#diff-1dc880f789fa7a9b157372ceb878d57d08af4ed374e5a50daba6ae9578ef3c4b)

Nice catch! Yes my bad, I probably copy-pasted from icub 3

closed 2023-02-16 by traversaro