robotology/icub-models #195: Right camera port does not open on iCubGazeboV2_5_visuomanip model

davidegorbani, 2023-03-30:
While I was doing the assignment https://github.com/vvv-school/assignment_make-it-roll I spawned the `iCubGazeboV2_5_visuomanip` model in Gazebo and noticed that the port `/icubSim/cam/right/rgbImage:o` corresponding to the right camera of the robot didn't open while the port `/icubSim/cam/left/rgbImage:o` the camera corresponding to the left camera opened. I report the output of `yarp name list`.

```
registration name /clock ip 10.240.9.59 port 10002 type tcp
registration name /clock/rpc ip 10.240.9.59 port 10003 type tcp
registration name /iCub/left_hand/analog:o ip 10.240.9.59 port 10005 type tcp
registration name /iCub/left_hand/analog:o/rpc:i ip 10.240.9.59 port 10004 type tcp
registration name /iCub/right_hand/analog:o ip 10.240.9.59 port 10007 type tcp
registration name /iCub/right_hand/analog:o/rpc:i ip 10.240.9.59 port 10006 type tcp
registration name /icubSim/cam/left/depthImage:o ip 10.240.9.59 port 10026 type tcp
registration name /icubSim/cam/left/rgbImage:o ip 10.240.9.59 port 10025 type tcp
registration name /icubSim/cam/left/rpc:i ip 10.240.9.59 port 10024 type tcp
registration name /icubSim/cam/right ip 10.240.9.59 port 10028 type tcp
registration name /icubSim/cam/right/rpc ip 10.240.9.59 port 10027 type tcp
registration name /icubSim/head/command:i ip 10.240.9.59 port 10013 type tcp
registration name /icubSim/head/inertials/measures:o ip 10.240.9.59 port 10029 type tcp
registration name /icubSim/head/inertials/rpc:o ip 10.240.9.59 port 10030 type tcp
registration name /icubSim/head/rpc:i ip 10.240.9.59 port 10012 type tcp
registration name /icubSim/head/state:o ip 10.240.9.59 port 10014 type tcp
registration name /icubSim/head/stateExt:o ip 10.240.9.59 port 10015 type tcp
registration name /icubSim/left_arm/command:i ip 10.240.9.59 port 10017 type tcp
registration name /icubSim/left_arm/rpc:i ip 10.240.9.59 port 10016 type tcp
registration name /icubSim/left_arm/state:o ip 10.240.9.59 port 10018 type tcp
registration name /icubSim/left_arm/stateExt:o ip 10.240.9.59 port 10019 type tcp
registration name /icubSim/right_arm/command:i ip 10.240.9.59 port 10021 type tcp
registration name /icubSim/right_arm/rpc:i ip 10.240.9.59 port 10020 type tcp
registration name /icubSim/right_arm/state:o ip 10.240.9.59 port 10022 type tcp
registration name /icubSim/right_arm/stateExt:o ip 10.240.9.59 port 10023 type tcp
registration name /icubSim/torso/command:i ip 10.240.9.59 port 10009 type tcp
registration name /icubSim/torso/rpc:i ip 10.240.9.59 port 10008 type tcp
registration name /icubSim/torso/state:o ip 10.240.9.59 port 10010 type tcp
registration name /icubSim/torso/stateExt:o ip 10.240.9.59 port 10011 type tcp
registration name /root ip 10.240.9.59 port 10000 type tcp
registration name fallback ip 224.2.1.1 port 10000 type mcast
*** end of message
```

xEnVrE, 2023-03-30:
I just noticed that the assignment is making the assumption that the port `/icubSim/cam/right/rgbImage:o` exists as well. Let me open an issue for that.

traversaro, 2023-03-30:
> I will also open an issue on the fact that the port names for the rgb images between the robot and the simulated robot are different, this is definitely not something expected and even if we do not fix the problem immediatly, we can use that kind as documentation.

Ah, but we have that: https://github.com/robotology/icub-models/issues/47 .

davidegorbani, 2023-03-30:
Hi @xEnVrE,

it is clear now, thank you for your help.

PR #196 (2023-03-31): Make iCubGazeboV_2_5_visuomanip RGB camera port names harmonized again
As per the title, this restore port names for the RGB cameras of `iCubGazeboV_2_5_visuomanip` such that they are as in 

https://github.com/robotology/icub-models/pull/45/files#diff-1e5c2f03288ab12a5b18118bc44ed0d0096511a21c7c2e02fe851c185e3de101

and not as in 

https://github.com/robotology/icub-models/pull/188/files#diff-e4397ef8e3fdcf273315c11bd778759f2352f511dbf53babc3c7b2b8284fa491

where the right camera port name was changed to `/icubSim/cam/right` by mistake.

With this change, the port names associated to the came [... cut]
closed 2023-03-31 by traversaro