robotology/icub-models #198: Disalignment between iCub 2.* models (such as iCubGazeboV2_5) and real iCub 2.* models regarding analogServer devices publishing information on /icub/<...>/analog:o ports

traversaro, 2023-04-05:
In https://github.com/robotology/icub-models-generator/pull/231 all the `analogServer` wrappers that were exposing FT sensors measures /icub<Sim>/<...>/analog:o ports were removed, and this change was released in icub-models 2.0.0 . See https://github.com/robotology/icub-models-generator/blob/fd2d430a399119f2ded5a15f1c8ea4185b9aca6c/simmechanics/data/icub3/conf/icub.xml#L23 .

 In their place, there are `multipleanalogsensorsserver` drivers that publishes information on a `/icub<Sim>/<..>/measures:o` ports, among other ports. 

However, physical iCub 2.* robots still just have `analogServer` wrappers to expose FT sensors measures, see:  https://github.com/robotology/robots-configuration/blob/b01aa43c2f51f477cf5b3c855a7e51ec46700ffb/iCubGenova11/icub_all.xml#L97 .

So, at the moment there is no way to write the same code that runs for both simulated and real robot, that has always been the main advantage of our simulators based on `gazebo-yarp-plugins`.

Possible fixes are:
* Add back `analogServer` devices to iCub 2.* simulated robots
* Substitute `analogServer` devices in real robots with `multipleanalogsensorsserver` devices

cc @Nicogene @pattacini @gabrielenava

Nicogene, 2023-04-05:
> Substitute analogServer devices in real robots with multipleanalogsensorsserver devices

I am for this option since @randaz81 will remove `AnalogServer` in future YARP releases
I can open a robots-configuration issue for that, since `AnalogServer` is still around, this cleanup has not to be included in 2023.02.2, but it is nice to have for future Distro

cc @pattacini

pattacini, 2023-04-05:
Agree @Nicogene 👍🏻

Nicogene, 2023-06-21:
This misalignment should be now solved am I right?
The `visuomanip` models have keept out from this refactoring because they don't expose FT sensors (if I am not wrong).

cc @pattacini @martinaxgloria

traversaro, 2023-06-21:
> The `visuomanip` models have keept out from this refactoring because they don't expose FT sensors (if I am not wrong).

I think they expose them: https://github.com/robotology/icub-models/blob/b209ab84411c6a64e297a3bc98146a2f576d7984/iCub_manual/robots/iCubGazeboV3_visuomanip/model.urdf#L3142 .

Nicogene, 2023-06-21:
> > The `visuomanip` models have keept out from this refactoring because they don't expose FT sensors (if I am not wrong).
> 
> I think they expose them:
> 
> https://github.com/robotology/icub-models/blob/b209ab84411c6a64e297a3bc98146a2f576d7984/iCub_manual/robots/iCubGazeboV3_visuomanip/model.urdf#L3142
> .

Good catch, I was looking into the `conf_manual` folder

Nicogene, 2023-10-04:
@xEnVrE you can take this as reference:
- https://github.com/robotology/icub-main/pull/905

traversaro, 2024-03-06:
I think we can close this. The last point mentioned here will be tracked in https://github.com/robotology/icub-models/issues/198#issuecomment-1745421875 , but it is not related to the problem described here.

closed 2024-03-06 by traversaro