robotology/icub-models #209: iCubGazeboV3 model feet contacts are unstable in Gazebo v11.13

gabrielenava, 2023-09-19:
I have Gazebo v11.13. after dropping the `iCubGazeboV3` model in the simulator, the robot starts to drift on the floor and the feet contacts are very unstable:

https://github.com/robotology/whole-body-estimators/assets/12396934/27f00d38-3fa0-4638-9cc4-c9b2be324ef5

this is not occurring in @HosameldinMohamed (previous) configuration, with Gazebo 11.12 (see also https://github.com/robotology/whole-body-estimators/issues/169#issuecomment-1716010096):

https://github.com/robotology/whole-body-estimators/assets/45564317/6ed58c8e-b97e-4389-8c24-7e23f0da1d0d

after he also updated to Gazebo 11.13, the same issue appeared on his laptop.

cc @traversaro

HosameldinMohamed, 2023-09-19:
Actually, I did more tests, and I found that probably updating Gazebo version is not the cause of this issue.

When I did my test with the latest Gazebo version, I used a clean Conda environment that also installed the latest of `icub-models` (`v2.3.0`).

The reason for this issue seems to be updating `icub-models` from `v2.2.2` to `v2.3.0`!

Which most likely is due to https://github.com/robotology/icub-models-generator/pull/243

@gabrielenava you can try to downgrade `icub-models` and check again.

traversaro, 2023-09-20:
Fix proposed in https://github.com/robotology/icub-models-generator/pull/251 .

traversaro, 2023-09-20:
@gabrielenava @HosameldinMohamed can you try with the latest devel, i.e. https://github.com/robotology/icub-models/commit/78b150c0441f09971a3da18fe542e624a779ee9d ? If it works, I can do a release.

HosameldinMohamed, 2023-09-21:
For me it works from the `devel` branch!

[image]

gabrielenava, 2023-09-25:
looks fine also for me! contacts are stable and the robot does not drift anymore!

[image]

traversaro, 2023-10-30:
Fixed released in https://github.com/robotology/icub-models/releases/tag/v2.4.0 .

closed 2023-10-30 by traversaro