robotology/icub-models #234: Migrate simulated MAIS interfaces to use MultipleAnalogSensors interfaces

traversaro, 2024-03-06:


traversaro, 2024-03-06:
As discussed in https://github.com/robotology/icub-models/issues/198#issuecomment-1745421875 .

traversaro, 2024-09-18:
@martinaxgloria  if you are working on iCubGenova11, can you report here:
* The ports on which iCubGenova11 is publishing mais data
* An exert of data published there (just saving a few lines of `yarp read ... /name/of/the/port` and reporting them here is enough, thanks)

traversaro, 2024-09-18:
@martinaxgloria can you also report here the patch you have been working with? Thanks!

martinaxgloria, 2024-09-18:
> @martinaxgloria can you also report here the patch you have been working with? Thanks!

The patch regards these two files:

- https://github.com/robotology/icub-models/blob/master/iCub_manual/conf_manual/iCubGazeboV2_5_visuomanip/gazebo_icub_left_hand_mais.ini
- https://github.com/robotology/icub-models/blob/master/iCub_manual/conf_manual/iCubGazeboV2_5_visuomanip/gazebo_icub_right_hand_mais.ini

In particular, I changed `name` and `device` under the `[WRAPPER]` group into:

- name: `/icubSim/*_hand/MAIS`
- device: `multipleanalogsensorsserver`

Here the two files:

```ini
[WRAPPER]
name /icubSim/right_hand/MAIS
period 10
networks (right_mais)
device multipleanalogsensorsserver

[right_mais]
jointNames r_hand_thumb_1_joint r_hand_thumb_2_joint r_hand_thumb_3_joint r_hand_index_1_joint r_hand_index_2_joint r_hand_index_3_joint r_hand_middle_1_joint r_hand_middle_2_joint r_hand_middle_3_joint r_hand_ring_1_joint r_hand_ring_2_joint r_hand_ring_3_joint r_hand_little_1_joint r_hand_little_2_joint r_hand_little_3_joint
```

```ini
[WRAPPER]
name /icubSim/left_hand/MAIS
period 10
networks (left_mais)
device multipleanalogsensorsserver

[left_mais]
jointNames l_hand_thumb_1_joint l_hand_thumb_2_joint l_hand_thumb_3_joint l_hand_index_1_joint l_hand_index_2_joint l_hand_index_3_joint l_hand_middle_1_joint l_hand_middle_2_joint l_hand_middle_3_joint l_hand_ring_1_joint l_hand_ring_2_joint l_hand_ring_3_joint l_hand_little_1_joint l_hand_little_2_joint l_hand_little_3_joint
```

martinaxgloria, 2024-09-18:
> * The ports on which iCubGenova11 is publishing mais data

`/icub/left_hand/MAIS/measures:o` and `/icub/right_hand/MAIS/measures:o`

> * An exert of data published there (just saving a few lines of `yarp read ... /name/of/the/port` and reporting them here is enough, thanks)

Here you are the output of `/icub/left_hand/MAIS/measures:o`:

```sh
() () () () () () () (((255.0 208.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.50703644753)) () ()
() () () () () () () (((255.0 207.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.52228999138)) () ()
() () () () () () () (((255.0 207.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.53259110451)) () ()
() () () () () () () (((255.0 207.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.54275965691)) () ()
() () () () () () () (((255.0 207.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.55237603188)) () ()
() () () () () () () (((255.0 207.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.56285738945)) () ()
() () () () () () () (((255.0 207.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.57274222374)) () ()
() () () () () () () (((255.0 207.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.58268594742)) () ()
() () () () () () () (((255.0 208.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.5931429863)) () ()
() () () () () () () (((255.0 208.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.60300111771)) () ()
() () () () () () () (((255.0 208.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.61310815811)) () ()
() () () () () () () (((255.0 208.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.62342000008)) () ()
() () () () () () () (((255.0 207.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.63314914703)) () ()
() () () () () () () (((255.0 207.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.64324045181)) () ()
() () () () () () () (((255.0 208.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.65355396271)) () ()
() () () () () () () (((255.0 208.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.66331982613)) () ()
() () () () () () () (((255.0 207.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.67343473434)) () ()
() () () () () () () (((255.0 208.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.68372559547)) () ()
() () () () () () () (((255.0 207.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.69350171089)) () ()
() () () () () () () (((255.0 207.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.70351862907)) () ()
() () () () () () () (((255.0 207.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.71407437325)) () ()
() () () () () () () (((255.0 208.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.72373723984)) () ()
() () () () () () () (((255.0 207.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.73379349709)) () ()
() () () () () () () (((255.0 207.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.74390625954)) () ()
() () () () () () () (((255.0 207.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.75407671928)) () ()
() () () () () () () (((255.0 207.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.7641723156)) () ()
() () () () () () () (((255.0 207.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.7742331028)) () ()
() () () () () () () (((255.0 207.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.78418135643)) () ()
() () () () () () () (((255.0 208.0 216.0 255.0 236.0 255.0 255.0 237.0 255.0 255.0 255.0 255.0 255.0 252.0 255.0) 1726646777.79436635971)) () ()
```

martinaxgloria, 2024-09-18:
Yes, sure. Here you are:

```sh
icub@iiticublap235:~$ yarp rpc /icub/left_hand/MAIS/rpc:o 
>>getMetadata 
Response: () () () () () () () ((id_x_hand_mais "" "")) () ()
```

traversaro, 2024-09-18:
Thanks! The revealed another bug (the names between left and right sensors is the same, creating problems if one wanted to insert the two sensors in the same remapper), but that is another story.

traversaro, 2024-09-18:
> the names between left and right sensors is the same, creating problems if one wanted to insert the two sensors in the same remapper

Reported in https://github.com/robotology/robots-configuration/issues/679 .

traversaro, 2024-09-18:
The fixes are provided in:
* https://github.com/robotology/gazebo-yarp-plugins/pull/688
* https://github.com/robotology/icub-models/pull/240

@martinaxgloria  could you test these PRs and check:
* If it solves your problem
* If data is streamed correctly on the `/icub/left_hand/MAIS/measures:o` port?

Thanks

traversaro, 2024-09-18:
> Hi @traversaro, thanks for the fixes!
> 
> I noticed that you named the [MAIS port](https://github.com/robotology/icub-models/pull/240/files#diff-a40b789602a26a5ae326b9ef6a5ccb3843ea9af343ccf274f81320a4cd070201) `/icub/*_hand/MAIS` and not `/icubSim/*_hand/MAIS`. Is it wanted?

No, it was a copy&paste error as I copied the file from the real world config files, fixed in https://github.com/robotology/icub-models/pull/240/commits/a33dd641ce8f3b14f54d225da99e513973907d12 .

martinaxgloria, 2024-09-18:
Now everything works properly, thank you!

Here the output of `yarp read ... /icubSim/left_hand/MAIS/measures:o`

```
mgloria@iiticb002lw003:~$ yarp read ... /icubSim/left_hand/MAIS/measures:o 
[INFO] |yarp.os.Port|/tmp/port/1| Port /tmp/port/1 active at tcp://10.0.2.73:10029/
[INFO] |yarp.os.impl.PortCoreInputUnit|/tmp/port/1| Receiving input from /icubSim/left_hand/MAIS/measures:o to /tmp/port/1 using tcp
() () () () () () () (((226.586270617047091491 240.806527604257126995 240.833359214207007426 226.581220191471402359 240.806274167312665213 240.83337727293496755 226.606676708060604142 240.813495979132653702 240.833385188201674509 245.503736883302707383 245.540381824538968658 245.555600494670756007 245.511500941734794878 245.542451678064082898 245.555598059000317335) 67.9549999999999982947)) () ()
() () () () () () () (((226.586916624035836776 240.806495663951153574 240.833391409820137596 226.581100451259260353 240.806291014289740815 240.833407692697477387 226.606514552808619101 240.813514386972855164 240.833399503177417955 245.503622039423134993 245.540395220722444947 245.555600815474861065 245.511387757124225573 245.542465507353284693 245.555596824004084056) 67.9650000000000034106)) () ()
() () () () () () () (((226.586480853462688856 240.806452985807737832 240.833351726132491422 226.581119565642211455 240.806344839434643745 240.833436224439850548 226.606750122021225025 240.813537488907201123 240.83343034157522311 245.504014705775233551 245.540389780659296548 245.555620846242248945 245.511793196675711215 245.542449959443729313 245.555612669975943163) 67.9749999999999943157)) () ()
```

```
mgloria@iiticb002lw003:~$ yarp rpc /icubSim/left_hand/MAIS/rpc:o 
>>getMetadata 
Response: () () () () () () () ((l_hand_mais "" "")) () ()
```

traversaro, 2024-09-18:
Great, thanks! I will merge and release all repos then.

pattacini, 2024-09-19:
Thanks heaps @traversaro!

traversaro, 2024-09-19:
@martinaxgloria the fixed version are now available in master branch of both gazebo-yarp-plugins and icub-models, and released in v4.12.0 and v2.8.0 (respectively). Updated conda binary packages will be available in around an hour (the time for the automatic process to start and end.

closed 2024-09-19 by traversaro

martinaxgloria, 2024-09-19:
Thanks a lot @traversaro!

traversaro, 2024-09-19:
(I just aligned devel of gazebo-yarp-plugins and icub-models with master branches).