robotology/icub-models #236: iCubLisboa01 - updated model

plinioMoreno, 2024-07-17:
Hi all,

We sent the iCubLisboa for upgrade during 2023, and it seems that the corresponding URDF model is not updated yet (current file [here](https://github.com/robotology/icub-models/commit/472db4555ea72832db571b49197849e1fc6192f1). @diogofbsilva wants to use the URDF corresponding to our robot, which I believe is version 2 on the arms and head, it is attached to a frame and has legs that move but do not walk (you can check the details with @andreaderito and @maggia80).

While you add the URDF to this repository, can you give us pointers on how to generate the URDF? I found some information [here](https://github.com/robotology/simmechanics-to-urdf), but it would be great if you let us know the current procedure to do it.

Looking forward to hearing from you.

diogofbsilva, 2024-07-18:
Hello everyone!

For now, if you are unable to send an updated URDF model of iCubLisboa, a model already generated that is more or less the same as our robot here would be extremely helpful as well.

Thank you!

maggia80, 2024-07-22:
@plinioMoreno @diogofbsilva you can find the URDF models here: https://github.com/robotology/icub-models

traversaro, 2024-07-22:
@plinioMoreno @diogofbsilva thanks for raising the issue, the iCubLisboa01 is currently generated with headV2, armsV2 and legs v1, see https://github.com/robotology/icub-models-generator/blob/1330dec35660f2e29a619b629abad4a71a97dd38/dh/CMakeLists.txt#L25 .

I asked internally and indeed the new version of iCubLisboa01 have the upperbody that is the 2.7 (so both the arms v2, but also the torso has the battery (i.e., its dynamics is different from a torso without a battery). Just to understand, are you using the legs of the robot at all? If yes, we need to understand how to merge a iCub 2.7 upperbody with a iCub 1.* lowerbody, otherwise it makes a lot of sense to just take a iCub 2.7 model if you are just using the upperbody.

traversaro, 2024-07-22:
If you do not use the lower limbs, you can use the `iCubGazeboV2_5_visuomanip` if you want to use the eyes and the hands, as your kinematics is the same of 2.5.5, 2.6.0 or 2.7.0 .

traversaro, 2024-07-22:
If you do not want to use the legs, you can just use the iCubGenova11 that is a v2.7 robot. We can probably prepare a iCubLisboa01 model that is a v2.7 with the legs removed, but in the meanwhile you can directly use the iCubGenova11 model.

plinioMoreno, 2024-07-22:
@traversaro and @maggia80 thanks for the pointers. We do not need the legs, since are just for aesthetic purposes. I believe @diogofbsilva can use already the iCubGenova11.

Let us know when the iCubLisboa01 is available in the repository.