robotology/icub-models #239: deploy_test_PyPI action broken

traversaro, 2024-09-06:
See https://github.com/robotology/icub-models/actions/runs/10738659245 . 

fyi @diegoferigo

diegoferigo, 2024-09-06:
The error is the following:

> ERROR    HTTPError: 400 Bad Request from https://test.pypi.org/legacy/          
         User 'traversaro' does not have two-factor authentication enabled.     
         Please enable two-factor authentication before attempting to upload to 
         PyPI. See https://test.pypi.org/help/#two-factor-authentication for    
         more information.

This is due to the change in PyPI authentication. I suggest to enable [Trusted Publishers](https://docs.pypi.org/trusted-publishers/) support as done in https://github.com/ami-iit/jaxsim/pull/174.

traversaro, 2024-09-06:
Thanks, I followed this on the test.pypi.org and it worked fine, thanks!

diegoferigo, 2024-09-06:
Keep in mind that you might also need to update the deployment GH workflow as done in https://github.com/ami-iit/jaxsim/pull/174. The action needs extra permissions.